JohnAtl/pcbgcode #1: Use small endmill to mill holes of various sizes

JohnAtl, 2018-05-26:
From Dave Z:
Hello. I love the pcbgcode add on but I can't use it. I have an ATC on my stepcraft 420 machine which is what I use for making boards but it only holds 5 bits and many of the parts I use have several different sized holes. What I have been doing to work around this is exporting the DXF file and importing it into VCarve. Then I use a 1/32 end mill and do a profile cut inside each hole.
I'm not much of a coder but I would think that this would be fairly easy enough to add and it would save having to have multiple drills in a rack as well as being able to cut even large holes for mounting etc.

MARIOBASZ, 2018-12-02:
This is good.
I think a good option is to choose one (or more than one) tool diameter to drill, and generate a helical descent of the target radius

MARIOBASZ, 2018-12-03:
I am after the development of my own economic ATC, so it is also a solution that interests me. With hole diameter and milling data, it is feasible to make larger diameter holes than the tool. I have made a complement for helical descent in BCNC, with help I think it can be implemented directly in Pcbgcode

deHarro, 2024-08-22:
Perhaps I'm wrong, but there exists a version of pcb_gcode with helical drill facility.
Please have a look at [this](https://github.com/JohnAtl/pcbgcode/tree/endmill-holes)...

I am using this version for a while now and I used helical drill to make holes of arbitrary sizes with my 1.4 mm drill bit.
Only drawback of this version is, the rounded corner feature of EAGLE is not correctly implemented.

These days I merged the fix for the rounded corner problem from V3.6.6-beta into this helical drill version V7.2 and it seems to perform neatly.

I'm not skilled in github pull requesting, so I don't think I should go ahead and try to implement those changes to Johns code on github.
But I found the github notification under the link I posted above:
[image]

I would interprete this so that all 21 changes to 3.6.6 ahead of the endmill-holes version would be integrated into the endmill-holes version, resulting in an all in one solution.

Harald