JohnAtl/pcbgcode #2: Milling with 3 bits

MARIOBASZ, 2018-12-03:
I John, here I show you what I have done using 3 bits: 0.10 - 0.7 - 3.0 and here I have implemented:
I have run pcb gcode 3 times, changing the dimension of the tool to 0.10, 0.8 (0.7 + 0.1) and 3.0.

[image]

[image]

[image]

[image]

In BCNC I have enlarged the traces of 0.1 in +0.7.
Intersect between traces of 0.7. Then I made a difference with the traces of 0.10. To the result perform pocket operation with 0.10

[image]

[image]

[image]

Between 0.7 and 3.0 the algorithm would have to be repeated (I have done it manually)

[image]

Perform a pocket operation on the dimensions of the board for bit 3.0, subtract the traces of 3.0 so that they do not enter it

[image]

JohnAtl, 2022-08-03:
Sorry I couldn't get to this. 
I'm assuming this has been resolved. Feel free to open a new issue if not.

MARIOBASZ, 2022-08-11:
Hi John
Not yet. I think this functionality would be useful in your pcb gcode.
I am finishing a plugin for bCNC that performs this operation (also useful for making posters, for example), it works on gcode blocks. I don't know programming, I'm very rough, but I'm close to finish it. It's a bit slow to deliver the tracks (35 seconds).
Regarding what you wrote about drilling with a diameter greater than that of the cutter, it is simple, and you can safely use G03 or G02 with R, in two semi-circles.
Scale-up for even larger diameters from a smaller diameter, can be with a discrete jump for each next diameter or an increasing spiral strategy.
Greetings Mario.