JohnAtl/pcbgcode #3: Mismatch between ULP and settings files versions - won't run

howiemnet, 2022-07-01:
The settings files (particularly pcb-machine.h) have been updated with new parameters, but the actual ULP hasn't. Looks like there was a new version of pcb-gcode made 4 months ago, but only half uploaded here.

Symptom: after a fresh install of this, when you launch pcb-gcode-setup from Eagle, it fails with "Unknown Identifier 'DEFAULT_WIDTH'. 

Nosing around it seems that DEFAULT_WIDTH is no longer in use - in older versions of pcb-gcode it was defined in pcb-machine.h, but that file doesn't have it.

The dates on the files in this current distro are a red flag - all the subfolders are new but the files in the root are ... really old.

JohnAtl, 2022-07-01:
Thanks for reporting this!
I'll try to have a look soon.

JohnAtl, 2022-07-01:
Should be resolved now. Please test. Thanks!

closed 2022-07-01 by JohnAtl