JohnAtl/pcbgcode #10: Using PCB-Gcode in Fusion 360?

TSJim, 2022-07-25:
I'm trying to use PCB-Gcode in Fusion 360.

I have installed PCB-Gcode 3.6.2.4 it into the same directory as the rest of the ULPs.

The good news is that the "pcb-gcode-setup.ulp" works fine.

However, when I try to run the "pcb-gcode.ulp" program, I get the following error message:

```
**/Users/jim/Library/.../examples/ulps/examples/pcb-gcode.ulp(1387):

polygons loop is depreciated.  Use polyPours polyShapes polyCutouts instead.**
```

Is there a way to fix this?

hilsonp, 2022-09-05:
An old post I made explaining my trial a few years ago might give you some ideas. 

https://groups.io/g/pcbgcode/topic/75874702#8182

I should give it another trial. Fusion360 fixed something, maybe.

JohnAtl, 2022-10-11:
Hi Jim,

You don’t have to run pcb-gcode directly, just set your options in pcb-gcode-setup, then click the Make my board (forget the exact wording) button. 
Give that a shot and see how it works.