JohnAtl/pcbgcode #8: corrections and addtions to pcb-gcode

deHarro, 2022-07-25:
Hi John,

seemingly you are on closing issues?

So perhaps you should have a short look at [my homepage](http://harald-sattler.de/html/pcb-gcode_-e-.htm).

There I elaborate on the "always climb" option and a really easy addition to your code to change the cutting direction of ALL pathes to always cut against the feed direction to get clean edges on traces as well as on the copper patches left over.
My patch is always activated, no checkbox in the GUI to have it switched on or off.

Another addition explained there is the patch suggested by Alexander Arkusha to get "spot drill" and "drilling holes", respectively,  working as expected.

Hope you see an advantage for your work.

Harry

JohnAtl, 2022-07-25:
Thanks Harry and Alexander!
I’ll have a look. 

I finished my PhD, so hope to have more time for hobbies now. 

—
JJ

mprowe, 2022-07-25:
I'd like to vote for the "Spot Drill" First Hole / Remaining hole fix. It's always catching me out!

Regards, M.

deHarro, 2022-07-25:
> I finished my PhD,

Congrats! 👍😃
Harry

JohnAtl, 2022-07-25:
> I'd like to vote for a "Spot Drill" First Hole / Remaining hole fix. It's always catching me out!
> 
> Regards, M.

Could you open an issue for that with a description?
I regret that I’ve lost track of issues and contributions people have made.

skybox53, 2022-07-25:
Congratulations on the educational accomplishment.  Very much admire the tenacity!

Roger

JohnAtl, 2022-08-10:
Thanks Harald!

closed 2022-08-10 by JohnAtl

deHarro, 2022-08-10:
You're welcome :)