JohnAtl/pcbgcode #9: When generating drill file, first drill OK. Remaing drills use "Spot Drill" depth?

mprowe, 2022-07-25:
Eagle Ver 9.6
PCB_GCODE ver 3.6.3
Repeatable: Yes.

Requesting drill only:
[image]
First hole is OK, subsequent holes use the "Spot Drill" depth. This is a snippet from drill file.

[image]

Workaround: Make the "Spot Drill" depth the same as "Drill Depth". And don't forget to return to correct value for "Spotting" while requesting Outlines.

Regards, M.

deHarro, 2022-07-27:
Hi M.!
You might try the fix from Alexander Arkusha explained on my [homepage](http://harald-sattler.de/html/pcb-gcode_-e-.htm).
The fix is described as second topic on this page.

Please keep an eye on the line numbers, perhaps you and I refer to different vesions of pcb-gcode.
But given the context, you will find the correct lines, where to change the code.

Harry

mprowe, 2022-07-27:
Hi Harry,
I'm not sure that the two problems are the same?
Using ver-3.6.3 (and previously ver-3.6.2.4) if I select "Generate bottom outlines" AND "Spot drill holes" I found no problem with the spotted points that were made. As far as I could tell the points were cut to the correct depth (0.2mm in my settings) by the V-Engraver after the track outlines were cut. i.e. both tasks were in the *.etch gcode file.

My problem, is different. If I select "Generate bottom (or top) drill" ONLY. The drill file will drill the first hole correctly but the rest will only be to the depth specified in the "Spot drill holes depth" field.

Lastly, just to be sure, I made the amendments described on your web page. As far as I could tell with the limited testing, I could detect no difference in the gcode output with or without these amendments! Now, it is highly likely that I have not done them correctly but I have attached them below in the hope that you could check them for me.

Regards, M.
[3files.zip](https://github.com/JohnAtl/pcbgcode/files/9200666/3files.zip)

deHarro, 2022-07-27:
Hi M.!
Your files seem been edited correctly to me.

As far as I remember, the problem addressed with Alexanders amendments, is, that if "spot drill" and "drill holes", respectively, are selected both, then the spot drills are executed with the depth defined for drilling the holes.

Whether these problems occur with two sided boards or only with one sided ones, I don't know. I didn't dig into it deeply.
I had those problems with one sided boards and the amendments of Alexander fixed the problem for me.

If your problem isn't solved with Alexanders amendments, then probably you found another problem in the code and we have to wait for Johns solution.

Harry

mprowe, 2022-07-28:
"... and we have to wait for Johns solution."

Good idea.

Regards, M.

JohnAtl, 2022-08-26:
@mprowe can you test one of the last versions I released and let me know if the problem is still there?
I can't seem to reproduce it.
Also, if you could send settings/*.h, that would be helpful too.
Thanks!