JohnAtl/pcbgcode #13: Milling the board outline

deHarro, 2022-08-11:
Hi John,

when I try to mill my board using the milling layer in EAGLE, the milling bit has no offset and the board becomes too small by half the cutter diameter.
Do I miss the place where the milling bit diameter can be specified, or is that an omission in pcb-gcode?
Thanks!

Harald

JohnAtl, 2022-08-11:
Set the line/wire width to the size of your cutter.
When you place the line in the milling layer, the cut path will be the line.
It's much easier for the human to do the tool offset, than pcb-gcode :-)

In the image below, I've set the line width to 1mm/0.039", and as I place the line, it shows where the cutter path will be. I should probably add this to the manual. Thanks for bringing it to my attention.

[image]

deHarro, 2022-08-11:
Ah, clean and easy :-) Thanks!
I was distracted from this possibility by the default line width of 0 (zero) for the outline of the board.
I will try that on my next board..
Harald

deHarro, 2024-08-26:
Hi @JohnAtl!

I comment on this already closed issue, since in the meantime I found a clever solution for the problem.

Let me elaborate on the original problem first.
Starting point is, I have a board to mill that has a somewhat cluttered perimeter.
Your proposal to change the line width and offset the cutting path manually is leading to success only in a feasible matter when the perimeter of the board is rectangular or at least has straight borders with little deviations from rectangular.

In the very example this is not given at all - have a look...

[image]

Those small cutouts with rounded ends and the diagonal corners would drive me mad if I where to form that manually.

So I searched for a tool which can help out and found [G-code Ripper](https://www.scorchworks.com/Gcoderipper/gcoderipper.html).

With this tool I convert the pcb-gcode generated "mill outline" file back into DXF format which can then be treated with e.g. [Estlcam ](https://www.estlcam.de/) to generate gcode again, this time respecting the tool diameter and even set holding tabs distributed on the perimeter of the board.

[image]

Perhaps a valuable hint for those who suffered from similar problems in the past... :)

Harald