JohnAtl/pcbgcode #19: Milling board outline with rounded corners

deHarro, 2022-09-02:
Hi John,

I tried to give rounded corners to my last design and used "miter" in Eagle to form the rounded corners.
The design looks exactly as expected in Eagle:

[image]

The board is double sided, the bottom layer is a GND plane with two 5 V wires.

Sadly, the milling doesn't perform as expected:

[image]

Look at those "ears" the milling process resulted in.
From the design I would have expected, that the rounding starts and ends in line with the adjacent straigth lines.

Is this a problem of pcb-gcode or does Eagle something strange?
Thanks!

Harald

JohnAtl, 2022-09-02:
You're right! Thanks for finding this.
I see this on the test board too.
A bug I introduced when switching from R to IJ for arcs.
I'll have a look.

[image]

deHarro, 2022-09-02:
> Thanks for finding this.

Youre welcome ;)
Thanks!
Harald

deHarro, 2023-10-02:
Hi @JohnAtl !

I just struggled over the same problem again and remembered, I raised a issue on that in the past.

Searching through your repo, I found a branch that seemingly has the solution to the described problem, so I eagerly copied the two changed files into my working solution of V3.7.0.
But as you surely would have expected, this was to no avail, since your changes on V3.7 are not present in the branch v3.6.6 with the remedies for rounded corners.
A short comparison of both sources with WinMerge shows, there is no easy merging both branches.

So, would you mind to do this (merging rounded corners into V3.7) in the (near) future?
I fear, without intimate knowledge of the code (as it is for me) I will have no success in trying that on my own.

Harald

deHarro, 2024-02-09:
Hi @JohnAtl !

Me again... :)
I have this issue still, sure, I didn't update pcb-gcode.ulp since then.

I found, you made V3.6.6 a release, so I assume, you wouldn't add the fixes for the milling layer problems on rounded corners into V3,7.0, would you?

No offense, just a question :)

Harald

deHarro, 2024-08-19:
Hi @JohnAtl !

I made another approach to that problem and this time I miracously had no problems to find and paste the changes for "rounded corners" into **my version** of pcb_gecode.ulp (V3.7.2_alpha).

I can not memorize my problems stated [here](https://github.com/JohnAtl/pcbgcode/issues/19#issuecomment-1742707694), but I think I managed to fix the "rounded corners" problem by pasting your changes to 3.6.6 into the version I have on my harddisk as version 3.7.2_alpha.
This alpha version has the option to make holes bigger than the drill bit by helical drilling.

So as of my point of view I may close this issue as fixed.
Please give a short comment. Thanks!

Harald

closed 2024-08-19 by deHarro